riscv/riscv-arch-test #452: Compliance command failure

T0288827, 2024-04-10:
I am trying to run compliance test for the Pico core using my UVM ENV but facing issue while running compliance command
 with reference to attached file.
[Compliance Testing Failure using the below command as follow.docx](https://github.com/riscv-non-isa/riscv-arch-test/files/14931382/Compliance.Testing.Failure.using.the.below.command.as.follow.docx)
 
Please let me know if any switch is missed to add in my compliance command.

MarcKarasek, 2024-04-10:
Clone this repo: https://github.com/InspireSemi/riscof_install_example_DUT

This repo is setup for using ACT to test a DUT vs Sail.  It has scripts to
install Sail and other dependencies needed by Sail/Riscof
There is a example_dut/ dir that has sample python scripts, plus DUT
environment headers.  You can use this as a template for your DUT.

You cannot just clone riscv-arch-test and run from there.  You need some
infrastructure around it that this repo provides.
If you have any questions about this repo email me and I can help you.

Marc

allenjbaum, 2024-04-15:
Did you try this fix?
  IF yes, did this fix your problem? 
      IF yes, please close this issue
  IF no, please describe what errors you get when following Marc's advice
      If no description is following, I'd advise the maintainer is to close this proactively in a couple of weeks.

T0288827, 2024-04-15:
Thanks for the reply,but for my project Riscof  is not the usage. Can you let me know is there any infrastructure to handle the test.S
Because I could see CVA6 and few more IPs are running without Riscof platform.

allenjbaum, 2024-04-15:
Sorry, but the arch-test sigs use the arch-test framework, which is riscof.
IF you don't use riscof, these tests won't help at all (since they are not self checking, and require running Sail and comparing the results)

There are also a set of spike tests that might be more appropriate for your use case:
https://github.com/riscv-software-src/riscv-tests
Caveat: they only support the configurations that Spike supports, and they don't have a lot of coverage.