riscv/riscv-arch-test #453: Unexpected Zfh's div test appear when no Zfh enable

wychlw, 2024-04-15:
When using the latest main branch with RISCOF with ISA `RV32IMFZicsr`, the `Zfh/src/fdiv_bx-01.S` tests (where x denotes every `fdiv` tests in `Zfh`) appeared in the test unexpectedly.

With some comparison between other tests in `Zfh`, I'm not fully sure but suspicious that the regex in check ISA is wrong, where originally be:

https://github.com/riscv-non-isa/riscv-arch-test/blob/8a0cdceca9f0b91b81905eb8497f6586bf8d1c6b/riscv-test-suite/rv32i_m/Zfh/src/fdiv_b1-01.S#L32

While in others like regex in mul is:

https://github.com/riscv-non-isa/riscv-arch-test/blob/8a0cdceca9f0b91b81905eb8497f6586bf8d1c6b/riscv-test-suite/rv32i_m/Zfh/src/fmul_b1-01.S#L32

It seems to be a similar problem as #448 

I tried to reproduce this problem using sail, with the following config: [sail_isa.yaml](https://gist.github.com/wychlw/c4d2d82e824490a98ba7f8c9ae0da83c)

wychlw, 2024-04-15:
Seems could be fixed with some simple changes, shall I make a pr for it if it's a problem?

allenjbaum, 2024-04-15:
Yes, please. *Every Zfh* test should require Zfh! (not just mul and div), though they seem to be the only ones that don't (besides Flh.align, which doesn't seem to even require F in the RVTEST_CASE, though it does in the RVTEST_ISA), which doesn't make sense unless it is unless used for FinX)

wychlw, 2024-04-16:
Fixed via #454

allenjbaum, 2024-04-17:
454 has been mreged, so closing this one

closed 2024-04-17 by allenjbaum