riscv/riscv-arch-test #456: Zfa tests missing fmvp.d.x

davidharrishmc, 2024-04-22:
PR #330 is pending to add support for Zfa.  The PR is missing fmvp.d.x, a RV32_Zfa instruction.

I suggest accepting the PR and adding fmvp.d.x later. This issue is to make sure the missing instruction is not forgotten.

Accodring to the PR #330  discussion (March 6, 2024), riscv-isac need enhancement to support fmv.p.d.

https://github.com/riscv-non-isa/riscv-arch-test/pull/330

There is a typo in the riscv-ctg PR (riscv_ctg/data/fd.yaml): s/fmvh.d.x/fmvp.d.x
After fixing the typo the test generation fails, because riscv-isac can't handle that instruction

allenjbaum, 2024-09-09:
ISAC PR#330 appears to be merged now. Can this issue be closed, or is something still needed?

davidharrishmc, 2024-09-10:
When PRs 489 and 490 are merged in, I’ll be able to run the flow and test.

closed 2025-02-10 by davidharrishmc