riscv/riscv-arch-test #459: Error in CTG commands for generating floating point riscv-tests

Shreesh-Kulkarni, 2024-04-24:
The command to generate floating point assembly tests in the fadd.d_b1-01.S file[https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/rv32i_m/D/src/fadd.d_b1-01.S](https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/rv32i_m/D/src/fadd.d_b1-01.S) does not work when invoked in the riscv-ctg repository via the corresponding cgf file [https://github.com/riscv-software-src/riscv-ctg/blob/master/sample_cgfs/sample_cgfs_fext/RV32D/fadd.d.cgf](https://github.com/riscv-software-src/riscv-ctg/blob/master/sample_cgfs/sample_cgfs_fext/RV32D/fadd.d.cgf).

The command in the comment is  missing some arguments(--base-isa) which had to be passed along with --flen instead of --xlen. 
Even after passing the arguments as requested by riscv-ctg, the following error pops up : 

[image]

allenjbaum, 2024-04-25:
shouldn't this be filed in the riscfv-ctg repo instead of arch-test?

Shreesh-Kulkarni, 2024-04-25:
I thought of posting it here as the command was explicitly stated in one of
the files in this repo. But I shall raise the issue in the riscv-ctg repo
as requested.

davidharrishmc, 2024-04-30:
See https://github.com/riscv-software-src/riscv-ctg/issues/111

davidharrishmc, 2024-04-30:
See also https://github.com/riscv-non-isa/riscv-arch-test/issues/400

jordancarlin, 2026-03-05:
riscv-arch-test has migrated to ACT4, which no longer uses CTG. Closing.

closed 2026-03-05 by jordancarlin