riscv/riscv-arch-test #460: Cover Group files are scattered in different directories

allenjbaum, 2024-04-24:
The .cgf files used to both generate test using riscv-ctg and to measure coverage of tests using riscv-isac are scattered around different repos: some in the riscv-ctg repo, some in the riscv-arch-test repo, and there may be others.

They should all be consolidated into the riscv-arch-test-repo

UmerShahidengr, 2024-04-26:
Rightly said. I will merge them into the riscv-arch-test repo and will include the rv32e and rv64e versions of it as well.

allenjbaum, 2024-04-27:
Just to be clear: I wasn't talking about the FP cgf files specifically, but
all of them.

Chris Muller, who is SW and Toolchain chair or vice chair, goes further.
He found a bug in ISAC C/I:  it uploads the tool to Pypi, but uses an
obsolete deprecated method to do it - so C/I now fails
Chris is of the opinion that riscof, ctg, isa, and riscv-config are all
part of arch tests, have no business being distributed to Pypi,
(they're not used separately) should be merged into the arch-test repo, and
the current repos should be archived.
I don't see why not, frankly, though a lot of init files might break, and a
lot of documentation will need to be updated.
(but a lot of documentation work is needed anyway)

This would solve problems of using out-of-date versions of one relative to
the other at least.

allenjbaum, 2025-05-02:
Everything is merged, no longer using Pypi

closed 2025-05-02 by allenjbaum