riscv/riscv-arch-test #478: 3.9.1 introduces problems with _start

olofk, 2024-07-06:
My regression tests for SERV started failing with version 3.9.1 of the arch tests with errors like `/home/runner/work/serv/serv/riscv-arch-test/riscv-test-suite/rv32i_m/I/src/add-01.S:1881: Error: symbol `_start' is already defined`. I tracked down the problem to commit 13252e1f3d4c871e13ebf4b6f2074972c599fb5d but I don't understand enough of the framework to meaningfully discern what to do.

Was this change intentional, and if so, is there anything I should do on my side to make it work again?

stnolting, 2024-07-06:
I am also using the latest version (without any specific adaptions for [3.9.1](https://github.com/riscv-non-isa/riscv-arch-test/releases/tag/3.9.1)) but I cannot confirm this bug on [my side](https://github.com/stnolting/neorv32-riscof). 🤔

> [...] with errors like _/home/runner/work/serv/serv/riscv-arch-test/riscv-test-suite/rv32i_m/I/src/add-01.S:1881: Error: symbol _start' is already defined_

That's strange. `_start` is not defined in that specific line/macro I think.

Did you add any custom software files to your port? `_start` might be defined in some core-specific start-up code (crto)?!

simonpcook, 2024-07-06:
I've done a bit of analysis on this looking at the source, and it looks like the underlying issue is with these two lines:

https://github.com/riscv-non-isa/riscv-arch-test/pull/469/files#diff-62959c4d982b131e0dbdf5ffd0d074f4f71c0c28cdf12ad6f68321f719d5b491R1880-R1881

I'm not sure why they're added, but it seems like it conflicts with (in the test you are running)

https://github.com/riscv-non-isa/riscv-arch-test/blob/dev/riscv-test-suite/rv32i_m/I/src/add-01.S#L24-L26

(I'm not quite sure why the linker error messages are wrong, but looking at your failing SERV CI test its the next line in the output)

The reason you are seeing it as a _start is that is how you define RV_MODEL_BOOT

https://github.com/olofk/serv/blob/main/verif/plugin-serv/env/model_test.h#L51-L55

(I'm assuming the intent of the RV_MODEL_BOOT macro isn't to be included multiple times in a compilation...)

olofk, 2024-07-06:
Thanks for the comments and analysis. I removed the RVMODEL_BOOT macro and things started behaving differently now. I haven't got it all figured out yet, but I started looking at @stnolting's support code for reference. The current SERV support code was ported over from the old framework and probably has a fair amount of unneeded legacy code in it.

olofk, 2024-07-06:
Alright. I got it working now by removing the RVMODEL_BOOT definition but keeping the define. Thanks all

closed 2024-07-06 by olofk