riscv/riscv-arch-test #493: misaligned jal[r]: write to link register?

stnolting, 2024-09-10:
If I understand the RISC-V spec. correctly, then an instruction that causes an exception must not _commit_. So if a `jal[r]` instruction jumps to a misaligned address it should not write the return address to the (link) register, right?

If this is true, then I think that particular part is not checked by the `misalign*-jalr` test cases?! 🤔

allenjbaum, 2024-09-10:
Hmm: the spec says"

The JAL and JALR instructions will generate an
instruction-address-misaligned exception if the target address is not
aligned to a four-byte boundary.
This is good, because it enables the trap handler to know where the odd
address came from, rather than just knowing that the it attempted to
execute at a failing address.
Note a corner case: if you set xEPC to an odd address, and return, it
should have behavior. This is a little nasty if it happens in Mmode since
it is irrecoverable.

So, to test whether or not a JAL[R] is committed, it merely needs to write
the link register into the signature. But, the link register is
linkerscript dependent,
so it needs to be adjusted (by storing a relative address instead of an
absolute address)
The JAL test does appear to do that; the last instruction of TEST_JAL_OP
subtracts the PC of the beginning of the macro, stores the link register
into the signature
That should determine whether or not the link register was set or not.
The JALR is more complicated, but appears to do the same thing in the end.

stnolting, 2024-09-15:
> The JAL and JALR instructions will generate an instruction-address-misaligned exception if the target address is not aligned to a four-byte boundary.

👍

> Note a corner case: if you set xEPC to an odd address, and return, it should have behavior. This is a little nasty if it happens in Mmode since it is irrecoverable.

So the address in `xEPC` can be unaligned causing an _instruction-misaligned_ exception. Aren't there any constrains defined in the RISC-V spec that - for example - `xEPC[1:0]` is hardwired to zero if IALING=32? I think that `xECP[1]` at least reads as zero if IALING=32?! 🤔

> So, to test whether or not a JAL[R] is committed, it merely needs to write the link register into the signature. But, the link register is linkerscript dependent,

I was not referring to `lr` specifically. To formulate my question more precisely: if a `jal[r]` instruction jumps to a misaligned address (and raising an exception) it should not write to the register file, right?

> The JAL test does appear to do that; the last instruction of TEST_JAL_OP subtracts the PC of the beginning of the macro, stores the link register into the signature

I'll have a closer look at this.

Thanks for clearing this!

allenjbaum, 2024-09-15:
I don't have time to look, but it may indeed be the case.
Note that IALIGN isn't necessary a constant (i.e.  MISC.C is writable ) and
the corner cases around changing it on the fly are messy
(and won't be handled by SAIL or ACTs because they involved
unpredictable behavior)

While the spec isn't explicit, the fact that JAL[R} generates an exception
implies that it doesn't retire, therefore no Xreg is updated.

closed 2024-09-17 by stnolting