riscv/riscv-arch-test #498: Error: illegal operands when trying to running floating-point testscase when using TEST_FPSR_OP

Pagerd, 2024-09-29:
I'm trying to add 'fcvt.s.h' testcases in act,Some parameters are as follows
```
inst_1:
// rs1 != rd, rs1==f29, rd==f30,fs1 == 0 and fe1 == 0x0d and fm1 == 0x1b7 and  fcsr == 0x0 and rm_val == 7  and rs1_nan_prefix == 0xfffff  
/* opcode: fcvt.s.h ; op1:f29; dest:f30; op1val:0x35b7; valaddr_reg:x3;
val_offset:1*FLEN/8; rmval:dyn; correctval:??; testreg:x2;
fcsr_val: 0 */
TEST_FPSR_OP(fcvt.s.h, f30, f29, dyn, 0, 0, x3, 1*FLEN/8, x4, x1, x2)
```

when using riscof running in riscof will showing error message

```
/home/pager/Desktop/work/ZfhTest/riscv-arch-test/riscv-test-suite/rv32i_m/Zfh/src/fcvt.s.h_b22-01.S:43: Error: illegal operands `fcvt.s.h f31,f31,dyn'
/home/pager/Desktop/work/ZfhTest/riscv-arch-test/riscv-test-suite/rv32i_m/Zfh/src/fcvt.s.h_b22-01.S:50: Error: illegal operands `fcvt.s.h f30,f29,dyn'
/home/pager/Desktop/work/ZfhTest/riscv-arch-test/riscv-test-suite/rv32i_m/Zfh/src/fcvt.s.h_b22-01.S:57: Error: illegal operands `fcvt.s.h f29,f30,dyn'
/home/pager/Desktop/work/ZfhTest/riscv-arch-test/riscv-test-suite/rv32i_m/Zfh/src/fcvt.s.h_b22-01.S:64: Error: illegal operands `fcvt.s.h f28,f27,dyn'
```

seems like `TEST_FPSR_OP` is not support fcvt.s.h

allenjbaum, 2024-09-30:
IS this a problem with the test, or a problem with the toolchain?

allenjbaum, 2024-09-30:
That is, a problem with the toolchain that you are using to run the tests.

jordancarlin, 2026-03-05:
The tests have all been rewritten for ACT4. Closing.