riscv/riscv-arch-test #500: Illegal Instruction exception due to overlapping registers

zeeshandildar, 2024-10-07:
I have encountered illegal instruction exception due to overlapping registers in vector crypto test i.e vsha2ms-e32.vv-01 (line 73). Replacing that with other register resolved the issue
```
inst_8:
TEST_CASE_WVV(4, 32, VINST, v16, 0*4, v17, 4*4, v16, 0*4)
//sig[120*4]
```

allenjbaum, 2024-10-08:
There should be some tests that check whether an illegal instruction
encoding would cause an exception. could this be one of them?
IF not, then we have to ask the developers why this test failure wasn't
caught in their testing of the code.

zeeshandildar, 2024-10-08:
I am not sure if this was intended to generate exception

allenjbaum, 2024-10-08:
We should probably consult with the author to confirm their intent before
accepting this PR, then

allenjbaum, 2024-10-08:
Oh, and if a trap was not the intent, we have to ask how this happened -
this should have been caught somewhere.
Another possibility is that this is optional behavior, and it was tested on
models that exhibit just one of them.

allenjbaum, 2024-10-16:
> I have found similar issue in other tests as well and in some cases it was
> causing load access fault as register gp (storing DATA_BASE) were
> overwritten by some instructions. I didn't create any merge request for
> that as I am not sure if it was intended or not. Here you can check that:
>
> https://github.com/riscv-non-isa/riscv-arch-test/blob/45b590cfc12e835da3c57e66dc1dfd5a7e6fbcb2/riscv-test-suite/rv32i_m/Zvk/src/vandn.vx-01.S#L116
> x3 is being used as source operand here but It was initialized to
> DATA_BASE, consequently producing load access fault.

My understanding is that none of these tests are expected to trap, so these
tests will likely need to be regenerated to avoid this case.
There should also be separate tests that check for this condition
(unless those already exist for other vector ops,  and we deem it
unnecessary to test it for all possible ops, which is probably the case).
We do need these tests for all vector op classes at least, e.g. .VV,.VI,
etc.

Generally (for base ops that can trap) these are put into the "privileged"
test suite, with the trap handler enabled.
We should figure out how to catch unexpected traps if the trap handler is
not instantiated (effectively installing a much simpler handler that does
nothing but end the test?).
If the trap handler is installed, we should shorten the default trap
signature area by a word.
That will cause the regular trap handler to end the test because it checks
for trap signature overflow.
Unfortunately, that only works if the trap signature area is defined to be
exactly the right size - and that isn't always possible,
as different implementations may trap or not depending on the architectural
options it supports).

UmerShahidengr, 2024-11-08:
Merged #501 to resolve this issue. Closing this one, but it can be opened again anytime

closed 2024-11-08 by UmerShahidengr