riscv/riscv-arch-test #518: Pip pinned package version dependency issues

jordancarlin, 2024-10-14:
PR #514 added many packages to the riscv-isac and riscv-ctg `requirements.txt` files and pinned them to specific versions. Many of these versions are fairly old, and other python packages depend on newer versions of these dependencies. Even within the riscv environment, riscv-config depends on `PyYAML >= 5.2` while riscv-isac requires `PyYAML == 5.1.1`. 

If these tools are both installed simultaneously using pip, then pip generates the following error message:
```
ERROR: Cannot install riscv-config and riscv-isac because these package versions have conflicting dependencies.

The conflict is caused by:
    riscv-isac 0.18.0 depends on pyyaml
    riscv-isac 0.18.0 depends on PyYAML==5.1.1
    riscv-config 3.18.3 depends on pyyaml==5.2

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict
```

Even in the CI logs for this repo, the following error is generated (though the CI still passes):
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
riscv-config 3.18.3 requires pyyaml==5.2, but you have pyyaml 5.1.1 which is incompatible.
pipx 1.7.1 requires packaging>=20, but you have packaging 19.0 which is incompatible.
httplib2 0.20.2 requires pyparsing!=3.0.0,!=3.0.1,!=3.0.2,!=3.0.3,<4,>=2.4.2, but you have pyparsing 2.4.0 which is incompatible.
```

jordancarlin, 2024-10-14:
The simplest fix is to switch the dependencies to be minimum required versions instead of exact versions, which I did in PR #519.

UmerShahidengr, 2024-10-14:
Nice work @jordancarlin .
Appreciated.

UmerShahidengr, 2024-10-14:
Approved your Pull request. Will close this issue once your PR is merged (after passing the CI)

jordancarlin, 2024-10-14:
Perfect. Thank you for the prompt response.

UmerShahidengr, 2024-10-14:
@jordancarlin there is one more thing which can be done i.e, to split rv32 and rv64 tests in 2 separate runs to achieve some parallelization. Are you available for running a quick update and making another PR?

jordancarlin, 2024-10-14:
Sure. I'll update the workflow matrix to run them as two separate jobs.

UmerShahidengr, 2024-10-14:
Great. Thanks

PR #519 (2024-10-14): Update requirements.txt to use >= required versions to allow for newer versions to be used
<FOR DOC UPDATES FILL ONLY DESCRIPTION AND RELATED ISSUES SECTION AND REMOVE THE OTHERS>

## Description

Update the `requirements.txt` files for riscv-isac and riscv-ctg to use minimum versions of dependencies instead or exact required versions.

### Related Issues

Fixes #518
closed 2024-10-14 by UmerShahidengr