riscv/riscv-arch-test #524: env path

JAYANTH-IITM, 2024-10-15:
Hi umer @UmerShahidengr 
this is the correct path to env : ~/riscv-arch-test/riscv-test-suite/env - (i passed this manual to get passed the error)
but if we try to generate a test using ctg , it pointing to the wrong path (the path that we specific for the test) ~/riscv-arch-test/riscv-test-suite/rv32i_m/zcmp/src/push.( this is a sample path i used for generating tests )

previously along with the each test that was generated using ctg the env was also copied there , but now since this was eliminated and everything this pointing to single arch_test.h , there is a error thrown up while generating using ctg .

error is from ctg.py 

Traceback (most recent call last):
  File "/home/saravana/.local/bin/riscv_ctg", line 33, in <module>
    sys.exit(load_entry_point('riscv-ctg', 'console_scripts', 'riscv_ctg')())
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/saravana/riscv-arch-test/riscv-ctg/riscv_ctg/main.py", line 30, in cli
    ctg(verbose, out_dir, randomize ,xlen, int(flen), cgf,procs,base_isa,inst,z_inx)
  File "/home/saravana/riscv-arch-test/riscv-ctg/riscv_ctg/ctg.py", line 110, in ctg
    if not os.path.exists(env_dir):
NameError: name 'env_dir' is not defined

UmerShahidengr, 2024-10-16:
@JAYANTH-IITM can you update this path, and make a pull request. We will merge this change right away. Thank you for pointing this issue.

JAYANTH-IITM, 2024-10-17:
ok sir will do it

closed 2024-10-18 by JAYANTH-IITM