riscv/riscv-arch-test #525: Add CI for codestyle

trdthg, 2024-10-16:
Hi, I want to add some CI to the codebase to check the code style and do some static analyze. I plan to use [ruff](https://github.com/astral-sh/ruff) , it will bring these benefits:

- The code style of the current repository is inconsistent and difficult to read, formatting can alleviate this. I will also appropriately modify the format config file based on the current code to make minimal changes.
- avoid some minor issues, such as duplicate keys in a dictionary
- Optimize the experience for other contributors, for example (if you use vscode and have installed the plugin, and sometimes you press ctrl-c, auto-save will automatically format the code, which is quite annoying)

Is any suggestions or comments?

trdthg, 2024-10-17:
for example thinks like these: https://github.com/riscv-non-isa/riscv-arch-test/pull/528

UmerShahidengr, 2024-10-17:
Thats a good suggestion @trdthg . Do you need any resources from our side to add changes in the CI, or will you update it yourself? If you are going to update it, then do add a PR, we will review it together and merge it.

trdthg, 2024-10-18:
Thank you, I'll send a PR later