riscv/riscv-arch-test #542: Need help on coverage mode test

trdthg, 2024-10-23:
I'm just running a very basic coverage test with the following command

```
riscof coverage --config=config.ini --suite=riscv-arch-test/riscv-test-suite/ --env=riscv-arch-test/riscv-test-suite/env \
    --cgf-file=./riscv-arch-test/coverage/dataset.cgf \
    --cgf-file=./riscv-arch-test/coverage/i/rv64i.cgf \
    --cgf-file=./riscv-arch-test/coverage/priv/rv64i_priv.cgf
```

_spike_isa.yaml_ just enable RV64I:

```yaml
hart_ids: [0]
hart0:
  ISA: RV64I
  physical_addr_sz: 56
  User_Spec_Version: '2.3'
  supported_xlen: [64]
```

but all the coverage are 0

[image]

Is this normal?

trdthg, 2024-10-31:
So anyone can help? The [riscof coverage documentation](https://riscof.readthedocs.io/en/stable/coverage.html#) does not seem to provide any useful information to help me debug

> My other PR #494 has also been stalled due to this issue

UmerShahidengr, 2024-10-31:
@trdthg, that's certainly not normal. Since your command is not giving any error, and doing exactly what its inputs are asking it to do, inputs to this command are sail log, database.yaml file, and MakeFile (generated automatically in riscof_work folder). The main issue can be in sail log, if possible, can you please check (and share) if sail log is getting generated correctly and tests are working fine?
In normal cases, tests are run on sail, its log is generated, and passed to ISAC for parsing. If the tests are not passing or sail log is empty, then ISAC will be getting empty files and will give no coverage.

UmerShahidengr, 2024-10-31:
Or there is another possibility as well, recently I posted an issue at sail related to its log format change ([details here](https://github.com/riscv/sail-riscv/issues/526)), in response to that issue, sail has added a command line argument called `--trace` ([more details here](https://github.com/riscv/sail-riscv/pull/528)), adding this command line argument in riscof plugin (just like I have added [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/2980588b5021d29a69596fc423c1b53fc088d672/riscof-plugins/rv32/sail_cSim/riscof_sail_cSim.py#L137)) will generate the sail log as required by ISAC.
Possibly you have not added this command line argument, and getting a wrong sail log format, which is causing this zero coverage issue.

UmerShahidengr, 2024-11-08:
@trdthg has your issue resolved?

trdthg, 2024-11-09:
I haven't had time to reconfigure my environment yet. I'll try to do it next week.

trdthg, 2024-11-13:
Thank you, after some attempts, I succeeded.
I directly created a new GitHub Codespace, installed Sail inside it, built the sail-model, and also installed riscv-toolchain and a bunch of other things, I got the correct coverage

This issue can be closed.

Then I am considering whether to create a workspace configuration file for this repository, which can help newcomers immediately start their work, or run a "hello world" of riscof

closed 2024-11-13 by trdthg