riscv/riscv-arch-test #552: [question] PMP tests only for m/s/u architectures?

stnolting, 2024-11-06:
Hi there!

I was trying the new pmp tests for a m/u-only architecture, but they keep failing because they use S-mode CSRs (`medeleg`, `satp, `sscratch` etc.) anyway. Is this intended or have I perhaps just overlooked a (new) setting?

If they are intended for m/s/u-architectures only then the ISA regex checks should be adjusted. For example

https://github.com/riscv-non-isa/riscv-arch-test/blob/fc32e41d49480fd99ba0a192dfff9c3319b44873/riscv-test-suite/rv32i_m/pmp32/PMP-CFG-reg.S#L39

should be extended to `check ISA:=regex(.*I.*S.*Zicsr.*)` (or even including U) in all PMP test cases.

allenjbaum, 2024-11-07:
I think you filed a pull request for that, and we'll get to it ASAP in the
dev branch.

UmerShahidengr, 2024-11-07:
Your PR is merged. Thanks for catching this one and fixing it.
Closing this issue now

closed 2024-11-07 by UmerShahidengr

stnolting, 2024-11-09:
Thanks!

Btw, are there any plans to generalize the PMP test cases so they can also be run on m/u and m-only platforms?

allenjbaum, 2024-12-23:
Since unlocked PMP entries are ignored by Mmode, tests for unlocked entires
should all rely on GOTO_LOWER_MODE macros to run the test in the
appropriate mode.
Those macros traverse modes by setting status.MPP to the lower mode, and
then returning with an MRET.
Because status.MPP is a WARL field, lowering to U-more will not lower the
mode in an M-only hart, so those tests will be run in Mmode - which may be
a bit redundant, but not incorrect.
Ditto for lowering to Smode.
In an M-U hart, lowerin to Smode is a bit more problematic, because you
don't know whether the test will be run in Mmode or Umode (welcome to WARL
freedom).
But as long as tests are designed to be run in all 3 modes, you can still
run the tests, and the ones that aren't applicable will just be redundant.

stnolting, 2024-12-23:
Thanks for clearing!

However, I think M/U-only support for the PMP tests is not "official" due to the regex filtering I've mentioned in the intial post (requiring `S` ISA extension):

https://github.com/riscv-non-isa/riscv-arch-test/blob/cd94912fed2aab75d7d5f115b441da0813fdce8d/riscv-test-suite/rv32i_m/pmp32/src/pmp-CFG-reg.S#L39

allenjbaum, 2024-12-25:
> @allenjbaum <https://github.com/allenjbaum> Thanks for clearing!
> Plus the test cases use CSRs like mdeleg that are not available on a
> m/u-only system.

Good catch. I agree; I think that the requirement for
defining rvtest_strap_routine must be conditional upon the presence of
Smode.