riscv/riscv-arch-test #566: I thought this is strange too.

trdthg, 2024-11-21:
After some testing, I found that the tests ctg (without `-r` option) generated each time is different. This is completely different from what I expected. I also had another person try it, and the results were also different.

I found that Tim should have found this problem a few month ago: https://github.com/riscv-software-src/riscv-ctg/issues/115

I am trying to fix this, but simply appending a line `random.seed(10)` after all` import random` statements does not solve the problem. I will check it again.

---
if you don't know what `-r` option does:

`-r, --randomize                 Randomize Outputs.`

> If randomization is enabled we use the `MinConflictsSolver` solver to find solutions.

https://github.com/riscv-non-isa/riscv-arch-test/blob/cd94912fed2aab75d7d5f115b441da0813fdce8d/riscv-ctg/riscv_ctg/generator.py#L493-L496

I haven't tested the behavior of `MinConflictsSolver`, which will use random inside

_Originally posted by @trdthg in https://github.com/riscv-non-isa/riscv-arch-test/issues/565#issuecomment-2488203492_

trdthg, 2024-11-26:
But I don't know how to handle random, according to the following code, it seems intentional ( I haven't looked closely at why to do this way)

https://github.com/riscv-non-isa/riscv-arch-test/blob/cd94912fed2aab75d7d5f115b441da0813fdce8d/riscv-ctg/riscv_ctg/cross_comb.py#L223-L225

allenjbaum, 2024-11-26:
While I wish it were reproducible, what is important is really the
coverage.
The only reason tests are recreated is to add specific cover points,

Test generation should not be adding tests that don't add new coverpoints,
so as
CTG runs, it must be keeping track of how much coverage has already been
found.

If we specify just new coverpoints, AND add the capability of pointing to
existing cover points
from a previous run, then new tests that match only existing coverpoints
could be dropped,
only tests matching new coverpoints will be added, and existing tests
won't remain unchanged.

That would make reviewing new tests much simpler.
I don't think that's an option that CTG has, though internally it must do
something like that when creating tests, so shouldn't be hard to add.
(I can say this because I know I'm incapable of making that change....:-

allenjbaum, 2024-11-26:
If you are generating tests for a specific extension, you can add a filter,
Riscof generates a test list default, so you could filter it, but  there is also an option to skip generating the test list and just test using an existing test list, which probably does what you want.

trdthg, 2024-12-03:
Some other research about random

Currently riscv_ctg will use random in the following places

- `misc/bitmanip_real_world.py`: It is a standalone script, and the seed has already been set inside
- `GeneratorCSRComb(base_isa, xlen, randomize)`: The parameters were passed, but not used
- `cross_comb.py`: There are many random calls here, but I didn't trigger them during testing, and I didn't find any cgf that used `cross_comb`
- `Generator`: Used to select `MinConflictsSolver`(which use random inside), if `-r` is not used, this is disabled

So currently, it seems that there is no need to modify any random seed, #570 is enough

allenjbaum, 2024-12-03:
Just to be clear: PR#570 will enable someone to add coverage, but not
change the existing test?

trdthg, 2024-12-04:
No, that will be another PR, #570 only fix reproducible when you rerun riscv_ctg

trdthg, 2024-12-04:
Oh, sorry I misunderstood your meaning, I thought you were saying that you are comparing with the existing test cases in the ctg repository currently.

> should it?

Although now I want to answer Yes immediately, I suddenly thought of a question:

If you for example

- Whether adding or reducing (including inserting and appending)
- Or just adjust the order

The results of the generated test cases may be different.

I did a test: taking sraw as an example, I added a new coverage point to its val_comb

```diff
    val_comb:
      'rs1_val == 0 and rs2_val >= 0 and rs2_val < xlen': 0
      'rs1_val < 0 and rs2_val > 0 and rs2_val < xlen': 0
      'rs1_val > 0 and rs2_val > 0 and rs2_val < xlen': 0
      'rs1_val < 0 and rs2_val == 0': 0
+      'rs1_val == (-2**(xlen-1)) and rs2_val >= 0 and rs2_val < xlen': 0
      'rs1_val == rs2_val and rs2_val > 0 and rs2_val < xlen': 0
      'rs1_val > 0 and rs2_val == 0': 0
      'rs1_val == (2**(xlen-1)-1) and rs2_val >= 0 and rs2_val < xlen': 0
      'rs1_val == 1 and rs2_val >= 0 and rs2_val < xlen': 0
```

结果是这样的

```patch
diff --git a/tests/sraw-01.S b/tests/sraw-01.S
index 549b42a1..a135e14e 100644
--- a/tests/sraw-01.S
+++ b/tests/sraw-01.S
@@ -2,7 +2,7 @@
 // -----------
 // This file was generated by riscv_ctg (https://github.com/riscv-software-src/riscv-ctg)
 // version   : 0.12.2
-// timestamp : Wed Dec  4 10:06:40 2024 GMT
+// timestamp : Wed Dec  4 10:06:56 2024 GMT
 // usage     : riscv_ctg \
 //                  -- cgf //                  --cgf /workspaces/riscv-arch-test/coverage/dataset.cgf \
 //                  --cgf /workspaces/riscv-arch-test/coverage/i/rv64i.cgf \
@@ -64,9 +64,9 @@ inst_5:
 TEST_RR_OP(sraw, x26, x25, x24, -0x1, 0x7fffffffffffffff, 0x1f, x1, 5*XLEN/8, x2)
 
 inst_6:
-// rs1==x24, rs2==x26, rd==x25, rs1_val < 0 and rs2_val > 0 and rs2_val < xlen, 
-// opcode: sraw ; op1:x24; op2:x26; dest:x25; op1val:-0xb504f332;  op2val:0x1f
-TEST_RR_OP(sraw, x25, x24, x26, 0x0, -0xb504f332, 0x1f, x1, 6*XLEN/8, x2)
+// rs1==x24, rs2==x26, rd==x25, rs1_val == (-2**(xlen-1)) and rs2_val >= 0 and rs2_val < xlen, rs1_val < 0 and rs2_val > 0 and rs2_val < xlen
+// opcode: sraw ; op1:x24; op2:x26; dest:x25; op1val:-0x8000000000000000;  op2val:0x1f
+TEST_RR_OP(sraw, x25, x24, x26, 0x0, -0x8000000000000000, 0x1f, x1, 6*XLEN/8, x2)
 
 inst_7:
 // rs1==x26, rs2==x25, rd==x24, rs1_val < 0 and rs2_val == 0, 
@@ -208,6 +208,21 @@ inst_34:
 // rs1_val == 1 and rs2_val >= 0 and rs2_val < xlen, 
 // opcode: sraw ; op1:x30; op2:x29; dest:x31; op1val:0x1;  op2val:0x1f
 TEST_RR_OP(sraw, x31, x30, x29, 0x0, 0x1, 0x1f, x9, 10*XLEN/8, x10)
+
+inst_35:
+// 
+// opcode: sraw ; op1:x30; op2:x29; dest:x31; op1val:-0x8000000000000000;  op2val:0x0
+TEST_RR_OP(sraw, x31, x30, x29, 0x0, -0x8000000000000000, 0x0, x9, 11*XLEN/8, x10)
+
+inst_36:
+// 
+// opcode: sraw ; op1:x30; op2:x29; dest:x31; op1val:-0x8000000000000000;  op2val:0x0
+TEST_RR_OP(sraw, x31, x30, x29, 0x0, -0x8000000000000000, 0x0, x9, 12*XLEN/8, x10)
+
+inst_37:
+// 
+// opcode: sraw ; op1:x30; op2:x29; dest:x31; op1val:-0x8000000000000000;  op2val:0x0
+TEST_RR_OP(sraw, x31, x30, x29, 0x0, -0x8000000000000000, 0x0, x9, 13*XLEN/8, x10)
 #endif
 
 
@@ -239,7 +254,7 @@ signature_x1_1:
 
 
 signature_x9_0:
-    .fill 11*((XLEN/8)/4),4,0xdeadbeef
+    .fill 14*((XLEN/8)/4),4,0xdeadbeef
 
 #ifdef rvtest_mtrap_routine
 tsig_begin_canary:
```

If you have added a new cover point, now given an input combination，ctg will try to find test cases that cover as many points as possible. Therefore, some things in inst_6 have been modified to cover more coverage points at the same time

However, these changes can be reviewed through the comments on the test cases, I think it's fine

So my answer is Yes now

> About adjusting the order: I have tested changing the order, but the generated result is the same. I think at least the **order of cover points in comment** should be different, maybe points has been sorted in some place, I don't know yet

UmerShahidengr, 2024-12-31:
Going to merge #570 well done @trdthg . Will close this one after merging the PR

PR #570 (2024-12-31): [CTG] Replace set with OrderedSet
## Description

resolve #566 

- Add new dependency `ordered-set`
	
  - I tried to implement an ordered-set myself, but I found it unnecessary; the ordered-set library is streamlined enough (only one file) and is fully compatible with the current usage of set (for example, overloading various operators)
- Replace `set` with `OrderedSet`
  
  - riscv-ctg use set to remove duplicates, but this will cause the unstable ordering of the for loop, and finally the generated tests will be different each time.

- Several very small if-else optimizations
  ```diff
  -if  [... cut]
closed 2024-12-31 by UmerShahidengr