riscv/riscv-arch-test #575: sd overwrites "nop; TEST_FOR_EXECUTION" as "c.unimp; c.unimp; nop"

marcfedorow, 2024-12-06:
https://github.com/riscv-non-isa/riscv-arch-test/blob/cd94912fed2aab75d7d5f115b441da0813fdce8d/riscv-test-suite/rv64i_m/pmp64/src/pmp64-NA4-R-priority-level-2.S#L52

This does that:

```
spike> mem 0x8000084c
0x00000013                                                             
spike> r 1
hart 0: 0x00000000800005c8 (0x00e7b023) sd      a4, 0(a5)
spike> mem 0x8000084c
0x00000000
```

I am not sure if this is intended.

allenjbaum, 2024-12-06:
I'm more than a little puzzled about what is going on here.
The comment says it is reading when it is actually writing, for one.
Without knowing a bit more about how the address parameter is used, I can't
tell what is going on; is it storing into the instruction stream?
But, if you're doing an SD (8byte), and the value in the source register is
0x13, why is spike reporting only 32b of the 64b of memory?

What is happening is that the source register has 0x13 in it, which (when
looking at it 32b at a time), is a nop in the low half, and 2 compressed
unimp in the high half.
And that's exactly what you're reporting is going on.

marcfedorow, 2024-12-06:
> I'm more than a little puzzled about what is going on here. The comment says it is reading when it is actually writing, for one. Without knowing a bit more about how the address parameter is used, I [...]

Spike is reporting as much bytes as I require (default=4).
I can ask for the whole line, i.e.:
```
spike> soc mem 0x80000840 +32
// before sd
80000840: 00000013 00000013 00000013 00000013 00000013 00140013 00000013 00000000
// after sd
80000840: 00000013 00000013 00000013 00000013 00000000 00000013 00000013 00000000
```

I agree with you that comments are providing negative amount of information.

This memory is being execuded a couple of instructions further:

```
spike>
hart   0: 0x000000008000084c (0x00000000) c.unimp
hart   0: exception trap_illegal_instruction, epc 0x000000008000084c
hart   0:           tval 0x0000000000000000
```

Instead of normally executing what should be execuded (i.e. `.word 00140013; .word 00000013`), it executes `c.unimp` and goes to trap.

The trap actually doesn't check a cause/etc so finally the test "successfully" finishes (with exit code 0), this is why this issue is not caught.

marcfedorow, 2024-12-06:
I might be wrong but for me seems obvious that WRX (in this order) should *not* overwrite anything (at least this is the intended behavior).
I.e. s[bhwd] writes what is already written just to check if write is allowed (but does not actually modify memory); l[bhwd] reads to check if read is allowed; and finally this instruction is being execuded to check that executing is allowed (this is what I could guess because of macro's name `VERIFICATION_RWX`).

allenjbaum, 2024-12-06:
Also, if it's storing into the instruction stream, it should require a
fence.i instruction.
If it executes the nop, followed by 2 unimplemented ops, it should trap
once, and it should store the cause into the trap signature (among other
things).
It will then return past the 2nd unimplemented op (again, that's what the
standard trap handler does) so won't trap twice.
Both of those aren't explicit in the test, that happens behind the scenes
in the standard trap handler.

marcfedorow, 2024-12-06:
> Also, if it's storing into the instruction stream, it should require a fence.i instruction. If it executes the nop, followed by 2 unimplemented ops, it should trap once, and it should store the [...]

I agree that it is what actually happens.
I just say that it seems obvious to me than it should not.

allenjbaum, 2024-12-06:
> I just say that it seems obvious to me that it should not.

Why? It tests that it can read (or not), it tests that it can write (or
not), and it tests that it can execute (or not).
That is the function of this test.

Since this is testing PMPs which have a granularity of 4 bytes, testing all
4 widths is probably overkill, but harmless

But it can distinguish between being allowed to execute  vs. executing an
unimplemented op by the cause CSR value (or by not trapping at all),
and by the EPC value (first or second word)
 All that information is captured in the trap signature, and if it matches
what Sail does, then the test works as (I perceive it to be) designed.

Why do you think this is incorrect?

I think the test should interleave writing and reading (and storing the
signature on every read)
instead of doing a bunch of writes followed by a bunch of reads and
signature store; you'll be testing for more bugs that way.
(coverage bug: it's missing a SIGUPD after LW)
(almost a bug:  its missing a */ after an extremely long comment that
should be wrapped to fit on a reasonably width screen),
  but luckily there is a comment that follows that enables it to be closed
You might consider it to be overkill to test 4 different sizes of stores
and loads (except word and double), but that's a different matter.

Still, the only functional bugs I see are:
 fence.i is missing, and
there are 2 copies of jal \ADDRESS in the macro definition, which will
cause the instruction after the macro invocation to be skipped

marcfedorow, 2024-12-06:
> Why? It tests that it can read (or not), it tests that it can write (or not), and it tests that it can execute (or not). That is the function of this test. Since this is testing PMPs which have a granularity of 4 bytes, testing all 4 widths is probably overkill, but harmless

No, it's not harmless.
Because of sd it doesn't actually test if it can execute from that region.
It will go to the trap in both cases: if it can't execute, it just can't execute; if it can, there are `c.unimp`s that are definitely not supposed to be there.

allenjbaum, 2024-12-07:
No, it does test it. The SD will store a real nop in the lower half, and
something that traps in the upper half. The EPC of the trap will not point
to the first half, but to the second half,
so the test will know if executed the first half, and execution is legal.

marcfedorow, 2024-12-08:
> No, it does test it. The SD will store a real nop in the lower half, and something that traps in the upper half. The EPC of the trap will not point to the first half, but to the second half, so the test will know if executed the first half, and execution is legal.

There are two reasons why that's wrong:
1. It's not reasonable to assume that the test is designed to check which cause leads to trap. It is obvious by design that `sd` should not store `c.unimp`s. For obvious reason such behavior is not implemented on any rv32 test that just normally executes this nop and checks if execution trapped. Implicitly writing `c.unimp`s is clearly not an intended behavior. Wasting hundreds of insns to write different causes via trap handler instead of just checking trap vs no trap (where "no trap" should be the correct path => not wasting insns) is definitely a bug, and the test works by luck (because trap handler doesn't know what causes are allowed so it just passes through).
2. Because of little endian, upper half (that contains two `c.unimp`s) is stored in lower addresses, i.e. test will **not** execute the "first half" because it is actually a second half, and the real first half are two `c.unimp`s (which is obvious from memory dump in my initial report).

marcfedorow, 2024-12-08:
By the way, it should check if it can execute `.word 0x00140013` from addr `0x80000850`.
By design `0x8000084c` must contain `0x00000013` (a.k.a. `nop`) and `0x80000850` must contain whatever it contains (e.g. `0x00140013`).
Because of `sd` overwriting `nop` by `c.unimp`s in address `0x8000084c`, test actually fails to check if it can execute from `0x80000850` which it should actually check.

allenjbaum, 2024-12-09:
Big endian is an interesting case.  We've tested nothing with big endian.
Instruction fetch is always little endian order, so lots of tests may
broken when data is big-endian and tests are mixing instructions and data.
But I'm not even convinced this is broken for big endian. Yes, it will trap
immediately on the jump -
but as long as you can tell the difference between an illegal instruction,
and an illegal address of an instruction - that's OK

But I completely disagree about most everything else. It really doesn't
matter what you think the intent is.
The intent, in this case, is to test whether it allows (or not allow)
reads&writes of different widths, and execution of memory that has been
marked as readable, writable, and executable (or not)
It doesn't matter if it takes a few hundred extra cycles - that is
optimizing something that is irrelevant for correctness, and nearly
unmeasurable for performance overall.
These tests aren't written for performance - that is not an intent (in this
case. For larger tests suite we might care)

What does matter is that it demonstrates the underlying cases it was
designed to exercise do get exercised, and that it does it correctly.
I would assert that it does exactly that, regardless of how elegantly it
does it.

I don't understand the comment about address 0x80000850  - that (and the
following) should all be nops,
so they should execute correctly even (assuming the text allows execution
at all, and if it doesn't, it means that the PMPs are set up correctly)
Therefore: not a bug, but not optimal .

As I have indicated, there are bugs: lack of fence, extra jal instruction.

marcfedorow, 2024-12-10:
> that (and the
following) should all be nops

True. And that is **not** the case, i.e. the following are not just nops, but also `c.unimp` that are irrelevant for the testing, do not belong to the address under testing and send execution to the trap for seemingly no reason — executing `nop` instead of `c.unimp` doesn't stop test from testing what it should; these `c.unimp`s exist only because it is an upper half of the register that contains `nop` and is being dumped to the PMP'ed `TEST_FOR_EXECUTION` by `sd`, which unpurposefully overwrites `TEST_FOR_EXECUTION +4` with the leading zeroes.
This is exactly what should be fixed, while I agree that the following should be `nop`s, not anything else.

allenjbaum, 2024-12-10:
Actually, only one unimplemented op will be executed; the trap handler will
skip over the following one. It always returns to the next doubleword after
the op that trapped.
I understand why those upper c.unimps exist, as do you. It ha
But let me be clear: redundant, unrelated, or unnecessary results don't
matter as long as the test hits its cover points and a DUT will match Sail.
We won't change the test just because it is ugly.

IF there is an actual test bug, such that a conforming DUT implementation
doesn't match Sail , or it doesn't cover a set of conditions we think we
should be testing- that's a different issue.

allenjbaum, 2024-12-25:
I'm going to close this issue because, as part of fixing real bugs, the value written into the executable will no longer zero-extended, but will contain a nop in the upper half. mIF this fails after the update is complete, this can br reopened.

closed 2024-12-25 by allenjbaum

marcfedorow, 2024-12-25:
I ask you to keep this open untill this issue is actually fixed in some new release.
It looks inconsistent in external documentation if it says "this patch is required to fix #575" but #575 is closed but is not actually fixed