riscv/riscv-arch-test #580: PMP tests assume 4-byte granularity

marcfedorow, 2024-12-18:
#577 is just a special case of this problem.
Tests that check NA4 / NAPOT and do not set 'X' flag just trap and never finish because nothing is allowed to execute from the region that contains the test's code.
Still, other tests' signatures don't match because of the wrong accumptions (i.e. 4-byte granularity).

allenjbaum, 2024-12-18:
I agree. The tests are being rewritten to take a granularity variable
(currently from the DUT specific RVMODEL_BOOT macro, I believe.
It will default to granularity of 4B if not defined in that macro.

marcfedorow, 2024-12-20:
I believe that granularity should be checked by software.
RISC-V priv spec says:
```
Software may determine the PMP granularity by writing zero to pmp0cfg, then writing all ones
to pmpaddr0, then reading back pmpaddr0. If G is the index of the least-significant bit set, the
PMP granularity is 2^(G+2) bytes
```

allenjbaum, 2024-12-20:
The granularity can be checked by SW, but the tests don't need to write &
read it  (except to verify that it matches what is claimed).
The tests can be compiled to match the granularity, and don't need case
statements to do different things for different granularity.
Yes, we could do that, but it's much simpler to write tests that compile
for the supported config (it's pretty much just a few of lines of code).