riscv/riscv-arch-test #588: Errata in PMP tests

davidharrishmc, 2024-12-31:
In at least pmp64-NAPOT-RWX.S and pmp64-TOR-RWX.S, I had to add a fence.i before the jal in VERIFICATION_RWX to get the tests working.  The store changes the instruction to a NOP, and the fence is necessary to fetch the correct instruction.

I see that pmp64-NA4-RWX.S has the same structure, and so do the pmp32 tests.  I suspect there might be something wrong with the signature generation such that no errors are flagged even without fence.i.

VERIFICATION_RWX also has several comments about Load Data that actually should be Store data.

jordancarlin, 2024-12-31:
The fence issue seems to be the same as #569

allenjbaum, 2024-12-31:
We do know about this issue, and it (along with several other issues) are
getting fixed. I'll leave this open until the fix is complete.
The tricky bit is making these tests work with arbitrary granularity.
Execute tests have to be placed at the end of a granularity, else returning
to the test keeps failing.
Large granularity will require large tests full of nops, and there is a
limit to the granularity size that tools allow.
WE are hoping a loader script cna fix this problem, by introducing an
arbitrary gap between the beginning and end of a granularity that doesn't
need to take up space in the elf file.
IF anyone has experience doing that sort of thing, get in touch with me or
Umer or Hammad.

davidharrishmc, 2024-12-31:
Confirmed that lockstep simulation of all 3 RWX tests in both pmp32 and pmp64 mismatch lockstep with ImperasDV even though only the two tests listed above cause signature mismatches.

Regarding NOPs, I had a similar issue in testing branch and jump offsets.  I ended up putting in enough NOPs to test up to 4k offsets, and assume jumps with an offset of 2^19 still work. Maybe 4k granularity is sufficient for PMP as well?

UmerShahidengr, 2024-12-31:
@davidharrishmc I agree with your comments, we are also targetting 4k grain limit while testing. Also on pmp32 and pmp64 mismatch, it is strange to see that rv64 picked this fence.i issue while rv32 failed to pick this one, we will look into it more detail.

allenjbaum, 2024-12-31:
What is the cause of the mismatch? Is it because Sail
does't support configuring granularity?

The problem isn't whether jumps work or not, it's whether the jump target
is executable or not(the tests should be trying cases where is is, and
where it isn't)
The problem with only testing 4K is that if the granularity is larger, then
you're branching into the middle of the larger region, and if execution
isn't allowed, it traps,
and returns to the region which still can't be executed, so you get
thousands of traps until it falls off the end into a different region (that
you hope is executable )

jordancarlin, 2024-12-31:
@allenjbaum The sail model has a flag (`--pmp-grain`) that configures the pmp granularity.

allenjbaum, 2024-12-31:
That's what I though, which begs the question as to why Imperas DV failed
the PMP test.
It could be for a known reason (fence.i is not executed prior to executing
a write into the instruction stream)

davidharrishmc, 2025-01-03:
ImperasDV failed all of the RWX tests because the tests don’t have fence.i.  Umer says this is a known issue being fixed.

allenjbaum, 2025-01-03:
yea, I pointed those out several weeks ago

davidharrishmc, 2025-04-13:
This seems to have been fixed and the test are working for me now.  I'm closing the issue.

closed 2025-04-13 by davidharrishmc