riscv/riscv-arch-test #590: Zcf/Zcd RVTEST_CASE macros inconsistent

jordancarlin, 2024-12-31:
The new Zcf and Zcd tests introduced in #497 and #587 are not consistent in their use of the RVTEST_CASE isa regex macro.

For rv32 and rv64 Zcd, c.fsdsp uses `check ISA:=regex(.*I.*D.*C.*)` while all of the other tests use `check ISA:=regex(.*I.*F.*D.*C.*)`. D implies F so this one isn't a big deal, but they should be consistent.

More significantly, for rv32 Zcf, c.fswsp uses `check ISA:=regex(.*I.*F.*C.*)` while all of the other tests use `check ISA:=regex(.*I.*F.*Zcf.*)`. Either approach is acceptable, but using `Zcf` requires modifying the ISA string that is passed to RISCOF when selecting tests. 

All of the Zcf and Zcd tests should use the same approach for this, either using the `Zcf` and `Zcd` extensions or the `C` and `F`/`D` extensions.

jordancarlin, 2024-12-31:
Related and even more problematic, the c.fldsp test for rv32i_m specifies rv64 instead of rv32 (`RVTEST_ISA("RV64IFDC")`)

jordancarlin, 2024-12-31:
The RVTEST_CASE check ISA regex is also missing 32/64 for rv32 and rv64 respectively.

jordancarlin, 2025-01-02:
@anuani21 @UmerShahidengr Is there a preference for one approach to the ISA string over another? I'm happy to open a PR to fix this, but we need to decide if we want that ISA string to use `C` or `Zcf`/`Zcd`.

UmerShahidengr, 2025-01-03:
Good catch @jordancarlin , I didnt check the ISA string for all tests, I think it should be consistent with Zcf/Zcd strings instead of C as these tests belong to the respective Zcf and Zcd extensions and not all C-extension enabled designs supports these tests

UmerShahidengr, 2025-01-03:
Would be great if you open up a PR to resolve this issue, we can do one thing, I can merge Zhix tests too, and @anuani21 and @jordancarlin can make all tests consistent via same PR. I am seeing some Zhinx tests also have the same issue

PR #597 (2025-01-16): Fix ISA strings for Zcf, Zcd, and RV64 sra
## Description

Fix inconsistent ISA strings in new Zcf and Zcd tests. Also restore `RV64` to sra ISA regex to avoid the following error:
```
ERROR | Error in test: /home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/rv64i_m/I/src/sra-01.S
Test Selected without the relevant extensions being available on DUT.
```

### Related Issues

@UmerShahidengr fixes #590
closed 2025-01-16 by UmerShahidengr