riscv/riscv-arch-test #602: Improving the README.md: Providing More Detailed Step-by-Step Command Instructions

LJY2000-max, 2025-01-20:
Your README.md file is terribly written. I followed the process you provided to install RISCOF, and under RISCOF, there are two components: riscv-ctg and riscv-isac. Why didn’t you include a step to first git clone this project and then use the command cd riscv-ctg inside the project’s folder?

Next, for the installation of Spike and SAIL, you similarly failed to specify which directory to perform the git clone in. How is anyone supposed to know where these cloned files should be placed?

Additionally, regarding the required environment files, there’s no explanation of how to write the config.ini file or where it should be located. For the riscv-test-suite and riscv-config, which are already present in this project’s directories, why are they even mentioned as needing to be cloned? Even if cloning is necessary, there’s no clarification on which directory to clone them into.

Lastly, when running tests, why didn’t you specify which directory the tests should be executed in?

In conclusion: This is an absolutely awful README.md file,I spent so many days to creating environment, but never succeeded.

allenjbaum, 2025-01-20:
I think we will have to plead guilty on all counts.
But, of course, resources: The original authors are gone,
and maintenance has been focused on just about everything else (primarily
the tests themselves, etc).
 if you would update the docs with clarifications that you think are
needed, file a PR and we will get them into the docs.

Arielfoever, 2025-04-07:
I can't agree you more.

allenjbaum, 2025-05-02:
can you close this if the latest docs meet your requirements/

Wren6991, 2025-05-07:
One remaining issue is that the command line in the readme to run the tests doesn't seem to be correct. Right now we have:

```bash
riscof run --config config.ini --suite riscv-test-suite/ --env riscv-test-suite/env
```

I instead had to run:

```bash
cd riscof-plugins/rv32
riscof run --config config.ini --suite ../../riscv-test-suite/ --env ../../riscv-test-suite/env
```

Pagerd, 2025-09-19:
I think the documentation is detailed enough.

allenjbaum, 2025-09-19:
If you think the read is detailed enough: evidence, please. What does the
readme say that wasn't followed by Luke.

There is a claim that the command line in the readme doesn't work.
If there is no other information in the readme that says which directory
level the command must be issued from, and it matters, then the readme
should be updated.

So please explain why you think it is detailed enough.