riscv/riscv-arch-test #609: [PMP tests] Does 'sfence-vma' needed after write pmpcfg register ？

ansyser, 2025-02-06:
When working in M mode and MMU is not enabled, is it necessary to execute the sfence-vma instruction with rs1=x0 and rs2=x0 to refresh the TLB after writing the PMP control status registers? The RISC-V manual describes the operation after configuring PMP registers as follows: "If page-based virtual memory is not implemented, memory accesses check the PMP settings synchronously, so no sfence.vma is need". How should we understand the phrase 'page-based virtual memory' in this sentence? Does it only refer to MMU, or does it also include TLB?

Taking "pmp64-TOR-RWX.S" as an example:
```
    /* Assigning addresses to PMP address registers */
    	LA(x4, PMPADDRESS0)		// Value to be stored in pmpaddr0
	srl x4, x4, PMP_SHIFT		// Shift right by 2 times
	csrw 	pmpaddr0, x4		// Updated pmpaddr0
	sfence.vma x0, x0     // need to add?
	nop				// Added nop in case of trap
```

allenjbaum, 2025-02-06:
TLBs only need refreshing if changes have been made to page tables I would
think.
Your example doesn't update any page table entries, and doesn't use the TLB
(unless status.MPRV is set).
My intepretation would be that "implemented" in this case also encompasses
bare mode and mmode without MRPV being set (I think there is a subtle
distinction between those two cases)

ansyser, 2025-02-07:
I found an issue in riscv-isa-manual, seems if our core support sv39/48，and working in m-mode, or in s-mode with satp.mode=bare, software needs to use an sfence.vma to sync PMP changes.

> Irrespective of what satp.mode one operates in, if sv39 is supported, that equates to "page-based virtual memory is implemented". And hence software needs to use an sfence.vma to sync PMP changes.
> 
> To help this make sense, note that designs that implement page-based virtual memory will have TLBs and will typically cache translations for all modes of operations. And, as part of caching translations, the associated PMA and PMP information will also be cached.
> 
> So, for Bare mode, trivial 1-to-1 address translations, along with associated PMA and PMP info, will be cached. In essence, Bare mode is not a special case for all the address datapaths.

https://github.com/riscv/riscv-isa-manual/issues/176

demin-han, 2025-11-11:
According to privilege 3.7.2 and https://github.com/riscv/riscv-isa-manual/issues/176, for S mode implemented core, sfence.vma is necessary after change PMP setting.
Currently, all PMP test cases have no sfence.vma.
Any plan to fix this?