riscv/riscv-arch-test #610: fence.i test and caches

algrobman, 2025-02-13:
Hi,
we see the test failure on CPU with Harvard architecture with D&I caches.
The test writes a new opcode with a store, executes fence.i and expects the new opcode to be executed.
However, the new opcode remains in the D-cache. To make it visible to I-Cache cbo.clean or cbo.flush is needed, which are not in the test code.
We see possible workaround as disable caching, but then we don't check that fence.i invalidates I-chache ...
On other hand, we want to run some other tests with caching ( for ex AMO tests only work on cacheable addresses).
How to select caching selectively per  test without patching their sources? (under riscof?)

What's the solution to run fence.i successfully on such CPUs?

allenjbaum, 2025-02-13:
The intent of fence.i is to make any prior data writes visible to the
fetch. If the fetch unit isn’t seeing a prior write after a fence.i, that
sounds  like a bug in your fence.i implementation. The fence.i should be
flushing dirty lines to the point of convergence- at least that’s my
understanding. It’s expensive, but it should work.
What does your fence.i do if not that?

algrobman, 2025-02-13:
fence.i does not effect D-side of CPU, but only I-side - instruction flow and invalidates I-cache. We had long discussions about this instruction spec and this was common agreement  ...