riscv/riscv-arch-test #640: Weird test execution when hypervisor extension is enabled

Allen784l, 2025-04-25:
Hello,
I am currently trying to execute RISC-V architecture test suite on a CPU supporting hypervisor extension/mode. 
In particular, I am running tests available under **rv64i_m/privilege/** directory. I am using Spike to generate the reference signature, as it seems that RISC-V Sail does not support **H extension**. The ISA specification I am using for **Spike** is **IMAFDCHZicsr_Zca_Zcb_Zba_Zbb_Zbc_Zbkc_Zbs**

Yet, although both DUT and reference signatures match on this group of tests, signature looks abnormal. In fact, due to H extension activation, there are some _unexpected_ remaining 0xdeadbeef in the signatures (that I do not see when Hypervisor extension is disabled). 

Taking **ecall** test from **rv64i_m/privilege/**  as an example, I am getting the next signature from Spike reference (notice the 0xdeadbeef at line 3-4):

[image]

However, if I run the same test on Spike reference, without Hypervisor extension (say, IMAFDCZicsr_Zca_Zcb_Zba_Zbb_Zbc_Zbkc_Zbs), the signature is:

[image]

To go more in depth into this strange signature difference, I have checked the instruction execution log from Spike, for both executions (with/without hypervisor extension), and one main difference is that execution with H extension enabled leads to trap overrun (under **chk_Mtrapsig_overrun** label, execution branches to **cleanup_epilogs**). There, no _mret_ is taken and execution stops, letting some 0xdeadbeef in the signature. 

Is that the expected behaviour of the test? Thanks for your help!

For more insights, here is a screen of the concerned execution divergence (left is Spike without Hypervisor, right is Spike with Hypervisor)

[image]

allenjbaum, 2025-04-25:
It will take me a while to look into this one.
As far as I know, no test has been set up to work with H-extension enabled.
Overrun means that it took more traps than the test was set up to do -
which is entirely possible, since the test isn't expecting hypervisor to be
enabled.
But also: the trap handler hasn't been tested with hypervisor enabled
either, and that could be the source of the problems.
I'd had to see a log of the execution to see what was going on.
The deadbeef signature isn't necessarily wrong, that's the background we
give the signature,
The signature region has two parts: the test signature, and the trap
signature. The text will explicitly store a value into the test signature,
and trap handler stores it into the trap signature.
There are also canary words before/after each subsection to catch
unexpected overwrites, and an overrun check as well.

Tests consist of a prolog which first initializes the model being test so
that it can run a test, carefully replaces any existing trap handler(s)
(one per priv mode) with our own, either by diverting traps to it,
or by copying a trampoline into where xTVEC points if xTVEC can't be
written properly (e.g. unknown alignment or range constraints)
It then initializes stack pointers and a memory map and sets up a save
areas (one per mode)
So, in this case, I'd expect 3 save areas be set up: Mmode, HS mode, and
VS mode, .
But this test shouldn't touch HS mode at all, it should operate entire in
Mmode, ECALLing into Mmode, so I'd need to see a trace to properly figure
out why it getting more traps than it should.

Allen784l, 2025-04-28:
Hello @allenjbaum, thanks for your help and the quick response. It gives me more understanding on what is happening. From the execution logs, I assume that the issue may be related to the size of the trap handler entry, that is larger when hypervisor is enabled (entry is 48byte-wide vs 32byte wide when H is OFF). 

[image]

Otherwise, I see no difference in execution until this overrun check is performed (in particular, trap exception is triggered once according to logs). I must have miss some initialization part in the testcode that set trap handler store region assuming 32byte-wide entry..

The only execution divergence I had, that occurs before the overrun check, is 

[image]

Allen784l, 2025-04-28:
I think I have spotted the issue. The trap signature size seems to be declared inside test assembly file *ecall.S*

[image]

If my understanding is correct, it indeed expects 32bytes of trap signature which is too low when hypervisor is enabled (48bytes are needed). It seems to be the cause of the overrun. I have checked by allocating 48bytes instead of 32 in the assembly file, and the signature looks good (no more 0xdeadbeef).

Allen784l, 2025-04-29:
Yes, according to test code (arch_test.h), this is due to the number of entries that are read on trap handler execution. When hypervisor is enabled, *tval2*, *tinst* registers are (expected to be) dumped, in addition to *epc*, *tval*, trap cause, and trap vector:

[image]

Trap handler is 6word signatures wide, when hypervisor is enabled, instead of 4.

I think it is this part of the test code that changes the required space for trap handler signature (and set it to 6word)

[image]

allenjbaum, 2025-04-30:
Aha, yes, that is the case. - The test wasn't expecting
hypervisor (misa.h==1) to be enabled.
This could be fixed by instead seeing if rvtest_vtrap_routine is defined
instead (or maybe to require both).

That would fix the problem also - but the trap handler needs to get fixed.
This should be retitled (or refiled) as "trap handler improperly stores
hypervisor signature in non-hypervisor test"

Thanks - this is a nice catch.

jordancarlin, 2026-03-08:
@UmerShahidengr @allenjbaum @Zain2050 I believe this issue has been resolved with the new trap handler. Can you confirm and close if so?

UmerShahidengr, 2026-04-15:
This issue was resolved with the new trap handler, and new self-checking tests of ACT4, so I am closing this one out, please feel free if you re-open it if you face this problem again.