riscv/riscv-arch-test #647: Example config requests excessive memory from spike

Wren6991, 2025-05-07:
I'm migrating from the old makefile flow to the new python flow. Following the tests in the readme, I tried testing spike against itself using the config in this repository (on the `main` branch). This line:

https://github.com/riscv-non-isa/riscv-arch-test/blob/b0179e138f18b85a8c79ea315c5cd256b65bab64/riscof-plugins/rv32/spike_simple/riscof_spike_simple.py#L161

Seems to request `1ull << 39` megabytes of memory from spike, per this output from `spike --help`:

```
Spike RISC-V ISA Simulator 1.1.1-dev

usage: spike [host options] <target program> [target options]
Host Options:
  -p<n>                 Simulate <n> processors [default 1]
  -m<n>                 Provide <n> MiB of target memory [default 2048]
  -m<a:m,b:n,...>       Provide memory regions of size m and n bytes
```

This causes spike to segfault. I'm not sure what the intent of that flag was, but things seem to run ok if it's removed.

allenjbaum, 2025-05-13:
...-m8796093022208 ...  - 8 Terabytes?
Separately: I object to using decimal values for memory size as opposed to hex values

Wren6991, 2025-06-19:
Thanks, looks like this was a mistake and is now fixed by #662.