riscv/riscv-arch-test #655: Compiling floating-point tests throws a bunch of bignum warnings

davidharrishmc, 2025-06-01:
GCC is now throwing tons of warnings about truncated bignums when compiling riscv-arch-test floating-point tests. I’m using GCC 14.2. I don’t recall these warnings earlier even though I’ve been on this version for a while. Jordan says the problem came up with a recent riscv-arch-test commit. The warnings can be reproduced with

```
cd /home/harris/cvw/tests/riscof/riscof_work/arch64/rv32i_m/D/src/fdiv.d_b9-01.S/dut; riscv64-unknown-elf-gcc -march=rv64ifd_zicsr          -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -g         -T /home/harris/cvw/tests/riscof/spike/env/link.ld         -I /home/harris/cvw/tests/riscof/spike/env/         -I /home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/env /home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/D/src/fdiv.d_b9-01.S -o my.elf  -DTEST_CASE_1=True -DXLEN=64 -DFLEN=64 -mabi=lp64
/home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/D/src/fdiv.d_b9-01.S:22630: Warning: bignum truncated to 8 bytes
```

An example of a troublesome line is

NAN_BOXED(0xbf800000000000003799029258263199744,64,FLEN)

jordancarlin, 2025-06-01:
I believe this is the issue same as #649.

Pagerd, 2025-09-16:
I think this issue has been fixed now