riscv/riscv-arch-test #657: PMP tests access medeleg even when S-mode is not implemented

Wren6991, 2025-06-03:
The PMP tests specify `def rvtest_strap_routine=True;`, like here in the CFG REG test:

https://github.com/riscv-non-isa/riscv-arch-test/blob/450de2eabfe4fcdfdf54135b5ab2dbb1d94805f8/riscv-test-suite/rv32i_m/pmp32/src/pmp-CFG-reg.S#L39

This inserts an access to `medeleg` here: https://github.com/riscv-non-isa/riscv-arch-test/blob/450de2eabfe4fcdfdf54135b5ab2dbb1d94805f8/riscv-test-suite/env/arch_test.h#L1001-L1005

However, `medeleg` is unimplemented on M,U-only machines, as per the spec:

[image]

Therefore this causes all PMP tests to take an illegal opcode trap during test initialisation if run on a machine which only implements M and U modes. PMP does not depend on S-mode (in fact it is the only memory protection mechanism available if you don't have S-mode) so this seems like a defect in the tests.

Experimenting locally with a couple of tests, it seems sufficient to just remove the `def rvtest_strap_routine=True;` from the test sources.

Wren6991, 2025-06-03:
Actually, looking closer at the other tests they do seem to use S-mode more extensively, so this is perhaps just a limitation of the tests.

allenjbaum, 2025-06-07:
IT's a bug; we overlooked M+U implementations. We have a fairly simple fix
for this,
though it does require changing all test_case instances everywhere that
define both rvtest_mtrap_ routine and rvtest_strap_routine with the same
condition to instead use separate conditions.
This can be done with a stream edit, so not too much work.

hirooih, 2025-06-08:
> we overlooked M+U implementations. 

We also overlooked M-only implementation for PMP tests.

***

And `medeleg` is not the only issue for PMP tests.

PMP tests have;

```
  csrw satp, x0			// Disable Address Translation
```

This  line must be removed (disabled) for implementations without S-mode.

***

And the most of PMP tests repeat same test in M, S, and U mode.  They should be conditional.  But this is not real issue 
 because `RVTEST_GOTO_LOWER_MODE` ignores unsupported mode. The test repeats the test in a same mode.

***

> define both rvtest_mtrap_ routine and rvtest_strap_routine with the same condition to instead use separate conditions.
> This can be done with a stream edit, so not too much work.

How do you do?

I found a discussion in #584.

> ?maybe something like? `RVTEST_CASE(1,"//check ISA:=regex(.32.); check ISA:=regex(.*I.Zicsr.); def rvtest_mtrap_routine=True; def TEST_CASE_1=True; def rvtest_strap_routine = check ISA:=regex(.S.);mac PMP_MACROS; mac PMP_helper_Coverpoints",pmp_cfg_locked_write_unrelated;)`.

But `def rvtest_strap_routine = check ISA:=regex(.S.);` causes an error.  It seems that it is not supported yet.

allenjbaum, 2025-06-10:
The fix we have also covers M-only configurations