riscv/riscv-arch-test #659: LA macro generates compressed instructions for platform that does not support C extension

object-Object, 2025-06-06:
I'm attempting to get riscv-arch-tests running on my [rv32ima emulator](https://github.com/object-Object/mlogv32). Currently, many/all tests are failing because the compiled binary includes `c.nop` instructions, even though the tests (and the toolchain) are compiled with the C extension disabled.

For example, the first test that runs with my ISA configuration is `rv32i_m/A/src/amoadd.w-01.S`. This is the output of `riscv32-unknown-elf-objdump --disassemble` for `inst_24` in this test:

```
000004a4 <inst_24>:
 4a4:	f80002b7          	lui	t0,0xf8000
 4a8:	fff28293          	addi	t0,t0,-1 # f7ffffff <immx+0x3d2f252f>
 4ac:	0c50a023          	sw	t0,192(ra)
 4b0:	fffe02b7          	lui	t0,0xfffe0
 4b4:	fff28293          	addi	t0,t0,-1 # fffdffff <immx+0x452d252f>
 4b8:	0c008393          	addi	t2,ra,192
 4bc:	0053a32f          	amoadd.w	t1,t0,(t2)
 4c0:	0c60a223          	sw	t1,196(ra)
 4c4:	00000013          	nop
 4c8:	00000013          	nop
 4cc:	00000013          	nop
 4d0:	00000013          	nop
 4d4:	00000013          	nop
 4d8:	00000013          	nop
 4dc:	00000013          	nop
 4e0:	00201417          	auipc	s0,0x201
 4e4:	cfc40413          	addi	s0,s0,-772 # 2011dc <signature_x8_0>
 4e8:	0001                	.insn	2, 0x0001
 4ea:	00000013          	nop
 4ee:	00000013          	nop
 4f2:	00000013          	nop
 4f6:	00000013          	nop
 4fa:	00000013          	nop
 4fe:	0001                	.insn	2, 0x0001
```

The values at `4e8` and `4fe` are `c.nop` instructions. This seems to correspond to `RVTEST_SIGBASE(x8,signature_x8_0)` in the test, which contains `LA(_R,_TAG)`. The `LA` macro seems to explicitly enable compressed instructions:

https://github.com/riscv-non-isa/riscv-arch-test/blob/450de2eabfe4fcdfdf54135b5ab2dbb1d94805f8/riscv-test-suite/env/arch_test.h#L489-L498

This issue(?) seems to have been introduced in https://github.com/riscv-non-isa/riscv-arch-test/commit/1da0f46a7a169645598df07933ea3dfde1805079. I manually reverted the changes made in that commit to `LA`, and the resulting build did not contain compressed instructions.

In case this is just a configuration error on my part, here are some details about my environment.

RISCOF DUT plugin files: https://github.com/object-Object/mlogv32/tree/4ee50d854509368c44846ff242fdac8c9e8c6f60/riscof/mlogv32

Compile command for the above disassembly:

```
riscv32-unknown-elf-gcc -march=rv32ia          -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -g         -T /workspaces/mlogv32/riscof/mlogv32/env/link.ld         -I /workspaces/mlogv32/riscof/mlogv32/env/         -I /workspaces/mlogv32/riscof/riscv-arch-test/riscv-test-suite/env /workspaces/mlogv32/riscof/riscv-arch-test/riscv-test-suite/rv32i_m/A/src/amoadd.w-01.S -o dut.elf  -DTEST_CASE_1=True -DXLEN=32 -mabi=ilp32
```

`riscv32-unknown-elf-gcc -v`:

```
Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-elf/15.1.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32ima --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 15.1.0 ()
```

The toolchain I'm using is available in [this Docker image](https://github.com/object-Object/mlogv32/pkgs/container/mlogv32-riscv-gnu-toolchain/428291305), which was compiled in [this workflow run](https://github.com/object-Object/mlogv32/actions/runs/15379905399).

object-Object, 2025-06-06:
Update: I was also able to get rid of the compressed instructions by adding `.option norelax` to `RVMODEL_BOOT`.

Maybe https://github.com/riscv-collab/riscv-gnu-toolchain/issues/445 is related?

object-Object, 2025-06-07:
Ok, I was able to get it to generate the compressed instructions in a standalone example, so it apparently isn't only an issue with riscv-arch-test.

`main.s`:

```
.global _start
_start:
    li x1, 0
    .option push
    .option rvc
    .align 5
    .option norvc
    li x1, 1
    .align 5
    .option pop
```

Commands:

```sh
riscv32-unknown-elf-gcc -march=rv32i -mabi=ilp32 -nostdlib -o main.out main.s
riscv32-unknown-elf-objdump --disassemble main.out > main.dump
```

`main.dump`:

```
main.out:     file format elf32-littleriscv

Disassembly of section .text:

00010080 <_start>:
   10080:	00000093          	li	ra,0
   10084:	00000013          	nop
   10088:	00000013          	nop
   1008c:	00000013          	nop
   10090:	00000013          	nop
   10094:	00000013          	nop
   10098:	00000013          	nop
   1009c:	00000013          	nop
   100a0:	00100093          	li	ra,1
   100a4:	0001                	.insn	2, 0x0001
   100a6:	00000013          	nop
   100aa:	00000013          	nop
   100ae:	00000013          	nop
   100b2:	00000013          	nop
   100b6:	00000013          	nop
   100ba:	00000013          	nop
   100be:	0001                	.insn	2, 0x0001
   100c0:	00000013          	nop
   100c4:	00000013          	nop
   100c8:	00000013          	nop
   100cc:	00000013          	nop
   100d0:	00000013          	nop
   100d4:	00000013          	nop
   100d8:	00000013          	nop
```

I tested this on the above toolchain, as well as a [similar one](https://github.com/object-Object/mlogv32/pkgs/container/mlogv32-riscv-gnu-toolchain/432220443?tag=rv32gc-ilp32) built with `--with-arch=rv32gc`. Both toolchains generated the compressed instructions.

I also tested it with the version of `riscv64-unknown-elf-gcc` that happened to be in the `gcc-riscv32-unknown-elf` package in the Ubuntu Focal repository. That did *not* generate the compressed instructions, but it seems to be a much older version of the toolchain that was compiled with very different options than mine:

```
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/9.3.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
gcc version 9.3.0 ()
```

Let me know if I should open an issue somewhere else (riscv-gnu-toolchain?).

allenjbaum, 2025-06-07:
Hmm, this has already been reported:
     https://github.com/riscv-non-isa/riscv-asm-manual/issues/84
I don't understand why the presence of .option norvc allows this; I've
asked.
It sounds like your solution of turning off relaxation will fix it, in any
case, but this is ugly unexpected(to me, anyway) behavior, and I' like to
see it fixed, unless someone can come up the a good reason for it. There
are a fair number of .align directives that would get affected by this.

object-Object, 2025-06-07:
Hm. Is that actually the same issue? Here's the intermediate object file for this example:

```sh
riscv32-unknown-elf-gcc -march=rv32i -mabi=ilp32 --compile -o main.o main.s
riscv32-unknown-elf-objdump --disassemble --reloc main.o > main.o.dump
```

```
main.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <_start>:
   0:	00000093          	li	ra,0
   4:	0001                	.insn	2, 0x0001
			4: R_RISCV_ALIGN	*ABS*+0x1e
   6:	00000013          	nop
   a:	00000013          	nop
   e:	00000013          	nop
  12:	00000013          	nop
  16:	00000013          	nop
  1a:	00000013          	nop
  1e:	00000013          	nop
  22:	00100093          	li	ra,1
  26:	0001                	.insn	2, 0x0001
  28:	00000013          	nop
			28: R_RISCV_ALIGN	*ABS*+0x1c
  2c:	00000013          	nop
  30:	00000013          	nop
  34:	00000013          	nop
  38:	00000013          	nop
  3c:	00000013          	nop
  40:	00000013          	nop
  44:	00000013          	nop
  48:	00000013          	nop
  4c:	00000013          	nop
  50:	00000013          	nop
  54:	00000013          	nop
  58:	00000013          	nop
  5c:	00000013          	nop
```

Unlike in the riscv-asm-manual issue, it seems like the assembler is inserting `c.nop` instructions, not the linker.

object-Object, 2025-06-07:
I think I might see what's happening. Here are the relevant functions in binutils:
* `riscv_frag_align_code`: https://github.com/bminor/binutils-gdb/blob/44e3c1c499331332412c1a04a348da2ddf9cf822/gas/config/tc-riscv.c#L5203-L5256
* `riscv_make_nops`: https://github.com/bminor/binutils-gdb/blob/44e3c1c499331332412c1a04a348da2ddf9cf822/gas/config/tc-riscv.c#L5175-L5196
* `riscv_handle_align`: https://github.com/bminor/binutils-gdb/blob/44e3c1c499331332412c1a04a348da2ddf9cf822/gas/config/tc-riscv.c#L5260-L5302

I think this is the sequence of events:
* The first `.align 5` calls `riscv_frag_align_code(5)` with rvc enabled.
  * `bytes` is 32, `worst_case_bytes` is 30.
  * `riscv_make_nops(nops, 30)` is called, generating one 2-byte NOP (and several 4-byte NOPs) because `bytes` is 30.
* The subsequent code is aligned to 2 bytes.
* The second `.align 5` calls `riscv_frag_align_code(5)` with rvc disabled.
  * `last_insn16` is true because of the previously emitted 2-byte NOP, so `frag_align_code(2, 0)` is called to align back to 4 bytes.
    * `excess = bytes % size` is 2, so even though `rvc` is disabled, `riscv_make_nops(p, 2)` is called and generates a 2-byte NOP to align back to 4 bytes.

allenjbaum, 2025-06-25:
If .option norvc will not prevent a c.nop from being generated, I'm not
sure what we can do.
This sounds like a toolchain bug.

UmerShahidengr, 2025-11-14:
Closing this one, as this seems fixed.

closed 2025-11-14 by UmerShahidengr