riscv/riscv-arch-test #665: PMP tests fail when TOR not implemented

fanghuaqi, 2025-06-17:
According to the spec here

https://github.com/riscv/riscv-isa-manual/blob/d124c244ffe1419589225c4e666a3bc652b4fdb8/src/machine.adoc#physical-memory-protection-csrs

> Up to 64 PMP entries are supported. Implementations may implement zero, 16, or 64 PMP entries; the lowest-numbered PMP entries must be implemented first. All PMP CSR fields are WARL and may be read-only zero. PMP CSRs are only accessible to M-mode.

PMP may only support NAPOT mode, but PMP cases such as vm_sv32 and pmp32/pmp cases require TOR mode, can I ignore the cases if PMP only support partial feature?

```
$ grep PMP_TOR -rn riscv-arch-test/riscv-test-suite/ | cut -d ":" -f 1 | sort -u
riscv-arch-test/riscv-test-suite/env/encoding.h
riscv-arch-test/riscv-test-suite/env/test_macros.h
riscv-arch-test/riscv-test-suite/rv32i_m/vm_sv32/src/pmp_check_on_pa_S_mode.S
riscv-arch-test/riscv-test-suite/rv32i_m/vm_sv32/src/pmp_check_on_pa_U_mode.S
riscv-arch-test/riscv-test-suite/rv32i_m/vm_sv32/src/pmp_check_on_pte_S_mode.S
riscv-arch-test/riscv-test-suite/rv32i_m/vm_sv32/src/pmp_check_on_pte_U_mode.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-CSR-ALL-MODES.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-NA4-M.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-NA4-S.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-NA4-U.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-NAPOT-M.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-NAPOT-S.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-NAPOT-U.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-TOR-M.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-TOR-S.S
riscv-arch-test/riscv-test-suite/rv64i_m/pmp/src/pmp64-TOR-U.S

```

allenjbaum, 2025-06-19:
That is because tests don't check whether TOR is implemented or not (or
because Sail can't be configured to match the behavior of the PMPCFG WARL
behavior)

But, I'm not sure what you mean by ignore, exactly. For self-certification,
you could ask for a waiver, and it should be granted.
You could also edit the test list to remove any test that has "-TOR-" in
its name.
(That won't help with eh CSR-ALL-MODES test; unclear if the "pmp_check_"
tests pass or fail right now.

allenjbaum, 2025-06-30:
I can't because you can't ignore it. You can either get mismatches with Sail and ask for a waiver because Sail doesn't implement the proper beavhior (doesn't support no TOR mode), or you can moddify the testlist to remove all TOR tests.
or you can wait until Sail has enabled removing the TOR encoding as legal.

fanghuaqi, 2025-06-30:
Hi @jordancarlin @allenjbaum, I have summited an issue about sail model missing PMP configuration, see https://github.com/riscv/sail-riscv/issues/1109.

UmerShahidengr, 2025-11-14:
All PMP tests have now been updated to rectify these issues, closing this issue, if there will be anyother problem, feel free to reopen this one or create a new issue.