riscv/riscv-arch-test #697: Error in make when test riscof example

CATEular, 2025-09-18:
ERROR | make: *** [/home/cat/riscv-arch-test/riscof-plugins/rv32/riscof_work/Makefile.Reference-sail_c_simulator:5: TARGET0] Error 1

which makes no Reference-sail_c_simulator.signature file.

jordancarlin, 2025-09-21:
More important than the version of the Sail compiler is the version of the RISC-V Sail model. What is the output of `sail_riscv_sim --version`?

CATEular, 2025-09-22:
> Can you attach the full output of running RISCOF?

Ok,sir.

cat@L:~/riscv-arch-test/riscof-plugins/rv32$ cd
cat@L:~$ source riscv_env/bin/activate
(riscv_env) cat@L:~$ cd riscv-arch-test/riscof-plugins/rv32
(riscv_env) cat@L:~/riscv-arch-test/riscof-plugins/rv32$ riscof run --config config.ini --suite ../../riscv-test-suite/ --env ../../riscv-test-suite/env

    INFO | ****** RISCOF: RISC-V Architectural Test Framework 1.25.3 *******
    INFO | using riscv_isac version : 0.18.0
    INFO | using riscv_config version : 3.20.0
    INFO | Reading configuration from: /home/cat/riscv-arch-test/riscof-plugins/rv32/config.ini
    INFO | Preparing Models
    INFO | Checked ISA file already exists
    INFO | Input-Platform file
    INFO | Loading input file: /home/cat/riscv-arch-test/riscof-plugins/rv32/spike_simple/spike_simple_platform.yaml
    INFO | Load Schema /home/cat/riscv_env/lib/python3.12/site-packages/riscv_config/schemas/schema_platform.yaml
    INFO | Initiating Validation
    INFO | No Syntax errors in Input Platform Yaml. :)
    INFO | Dumping out Normalized Checked YAML: /home/cat/riscv-arch-test/riscof-plugins/rv32/riscof_work/spike_simple_platform_checked.yaml
    INFO | Generating database for suite: /home/cat/riscv-arch-test/riscv-test-suite
    INFO | Database File Generated: /home/cat/riscv-arch-test/riscof-plugins/rv32/riscof_work/database.yaml
    INFO | Env path set to/home/cat/riscv-arch-test/riscv-test-suite/env
    INFO | Running Build for DUT
    INFO | Running Build for Reference

   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zcmop/src/c.mop.1-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zcmop/src/c.mop.11-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zcmop/src/c.mop.13-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zcmop/src/c.mop.15-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zcmop/src/c.mop.3-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zcmop/src/c.mop.5-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zcmop/src/c.mop.7-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zcmop/src/c.mop.9-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.0-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.1-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.10-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.11-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.12-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.13-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.14-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.15-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.16-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.17-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.18-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.19-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.2-01.S
Test Selected without the relevant extensions being available on DUT.
   ERROR | Error in test: /home/cat/riscv-arch-test/riscv-test-suite/rv64i_m/Zimop/src/mop.r.20-01.S
[... 69 lines cut ...]
  File "/home/cat/riscv_env/lib/python3.12/site-packages/click/core.py", line 1830, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  File "/home/cat/riscv_env/lib/python3.12/site-packages/riscof/framework/test.py", line 458, in run_tests
    result, diff = compare_signature(res, ref)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cat/riscv_env/lib/python3.12/site-packages/riscof/framework/test.py", line 46, in compare_signature
    file2_lines = open(file2, "r").readlines()
                  ^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/cat/riscv-arch-test/riscof-plugins/rv32/riscof_work/rv32i_m/A/src/amoadd.w-01.S/ref/Reference-sail_c_simulator.signature'

Zain2050, 2025-09-22:
It seems that there's probably a change in Sail that hasn't been made to the plugin file yet. 
sail_riscv_sim version 0.7 is working fine for me, maybe you can try that?

PR #698 (2025-09-30): (CI Fix) Disabled Svrsw60t59b extension from RV32 Sail plugin
Closes #697. 

The cause of failure was: 
**Svrsw60t59b** is enabled but Sv39 is disabled, supporting Svrsw60t59b requires supporting Sv39.
Therefore, I have disabled it from the RV32 plugin file.
["mtval_has_illegal_instruction_bits"] has a default value of True now, therefore that line in RV64 is no more required.